python-microscopy/python-microscopy #1412: [SUPPORT] Accesing 'MainFrame' from Other PYME Functions

Yujin-Bao, 2023-08-11:
**What are you trying to do?**
This has been solved in a way, but may need more attention in the future.
I am trying to crop a selected ROI in a funtion by using `MainFrame.OnMCamRoi(event=0)`. However I could not access MainFrame from other functions. The only way I could do so was to run it in the shell. It seems `OnMCamRoi` is not well separated from GUI code.
[image]
[image]

**The way I solved it**
Thanks for the hint offered by Andrew. I was able to solve it in a way. I added an init_gui section in the init script:
```
@init_gui('JustInTimeROI')
def JustInTimeROI(MainFrame, scope):
        import weakref
        scope.crop = weakref.WeakMethod(MainFrame.OnMCamRoi)
```
In this way I was able to run `scope.crop()(event=0)` to crop an ROI.

David-Baddeley, 2023-08-12:
This needs **way** more context ... where are you trying to run this (what file is the top snippet in) and what are you trying to accomplish? `MainFrame` is available as a convenience in the shell (for debugging) and as a parameter to `init_gui` script (so that gui controls can be added), but is not available elsewhere and shouldn't be.

Calling the GUI menu function handler to set the ROI is also going to be **super fragile** and could have unintended consequences - it's definitely not the recommended way of doing things! If you want to set the region of interest, the best way would probably be to interact directly with `scope.state['Camera.ROI']`.

I'd also strongly caution against modifying core functionality such as `xyzct.py` - especially by adding stuff which is specific to one hardware setup. Doing so will mean that you will not be able to cleanly update in future, and you will be stuck on old code. If you want to do something at the start of an acquisition, use a protocol. If you want protocol-like customisation for a standard z-stack, then we probably need to hack some form of pre-run hook in, but it will need to be generic and to work (or safely get out of the way) on all hardware.

Yujin-Bao, 2023-08-12:
Thanks for the reply, David! For acquiring SMLM I usually crop to a small ROI for drift tracking in the transmitted-light channel, so I am trying to automate the following: whenever I click 'Acquire Stack' in the OI-DIC GUI panel, the program will disable the drift tracking, turn the AOTF off, uncrop the ROI, capture OI-DIC frames, crop the ROI back, turn the AOTF back on, and enable the drift tracking. In this way I can capture OI-DIC at the same time of acquiring SMLM, which is more user-friendly and more drift stable since this saves time by only clicking one button.

The top screenshot is exactly `xyzct.py`. I did this because it seems it is only used in my OI-DIC acquisition for now, and I modified it in a newly created local branch on the scope computer. I agree this is pretty much not elegant... I wrote an acquisition protocol for this but I had no idea how to use it yet. I only see I can select a protocol in Time/Blinking series panel and whenever I select one it begins to run without any clicking. I was wondering if you had a hint about how to impletement this protocol selection in OI-DIC panel?