python-microscopy/python-microscopy #1425: [ENHANCEMENT] check if some warnings, info messages are still required/useful

csoeller, 2023-09-18:
**Description**

There are a few log/info messages that would be good to check if these are really still useful/needed.

There are two instances I am aware of:

1. I still get regular `DictMDHandler` warnings, would be good to check if these are still required, see also #1352 
2. Every CSV import gives `TextFileSource-use_pandas` info messages, which are slightly broken, too

**Use case where these are annoying**

These show up prominently in Jupyter notebooks when I import CSVs into pipeline objects, e.g. as shown below, where I often import a bunch of series. They mess up the output quite a bit. I know I can supress warnings, but ideally if these warnings are meaningful it would be good to know if something still needs fixing? 

Similarly, printing the info on `TextFileSource-use_pandas` each time seems not the optimal solution.

[image]

David-Baddeley, 2023-09-18:
Incomplete list of where `DictMDHandler` might still cause problems below (metadata entries accessed as attributes):

PYME/experimental/pointFitConfoc.py
IO/FileUtils/h5-thumbnailer
PYME/localization/ofind_pri
PYME/localization/ofind_xcorr
PYME/localization/ofind_xcorr_fw
PYME/localization/remFitBuf (ie all fitting)
localization/FitFactories/InterpFitR, MultiviewInterpFitNR, PRInterpFitQR, PRInterpFitR, SplitterFit*, baseInterpolator, all the estimators, fitTestJigs

I'm trying to fix these as I find them, but for now the warning is definitely warranted.

As to the pandas message, I think this is still semi useful as pandas import is broken for some text file formattings. I should make sure the string substitution works though!

David-Baddeley, 2023-09-18:
I think I've hit most of the obvious cases where we have assumed a `NestedClassMDHandler`, using the following regexes (which might be useful when looking at plugins)

`md\.(?!get)(?!set)(?!keys)(?!copy)(?!voxelsize_nm)(?!to_JSON)(?!json)(?![\s\n])`
`mdh\.(?!get)(?!set)(?!keys)(?!copy)(?!voxelsize_nm)(?!to_JSON)(?!json)(?![\s\n])`
`metadata\.(?!get)(?!set)(?!keys)(?!copy)(?!voxelsize_nm)(?!to_JSON)(?!json)(?![\s\n])`

It turned out to be pretty much all the fit modules and quite a few more besides.

I think the few that remain are all in comments, but due to the scope of #1427 I'm not 100% sure.

csoeller, 2023-10-02:
Many thanks for looking into it. I take the point about the `DictMDHandler`.

The `TextFileSource-use_pandas` message is probably of value as well as you mention.

I would argue against a simple `print` function though for any kind of user message and rather suggest that all messages should be logging or warning messages that allow filtering in some fashion. I suggest below a possible patch that would just do that. One can probably discuss at what level this should be, info or warning. I chose warning since then they will show up by default while info level messages may by default not be seen.

I have not yet raised this as a PR, but am happy to unless some major objections to this approach are raised.

```diff
diff --git a/PYME/IO/tabular.py b/PYME/IO/tabular.py
index 888d0de42..ce630e614 100644
--- a/PYME/IO/tabular.py
+++ b/PYME/IO/tabular.py
@@ -578,7 +578,7 @@ class TextfileSource(TabularBase):
         #print('invalid_raise:', invalid_raise)
 
         if config.get('TextFileSource-use_pandas',False):
-            print('Opening %s using pandas (set TextFileSource-use_pandas: False in config.yaml to use legacy np.genfromtxt instead)')
+            logger.warn('Opening "%s" using pandas (set TextFileSource-use_pandas: False in config.yaml to use legacy np.genfromtxt instead)' % filename)
             import pandas as pd
             self.res = pd.read_csv(filename,
                 comment=comments,
@@ -592,7 +592,7 @@ class TextfileSource(TabularBase):
                 ).to_records(index=False)
 
         else:
-            print('Opening %s using np.genfromtxt (set TextFileSource-use_pandas: True in config.yaml to use pandas instead)')
+            logger.warn('Opening %s using np.genfromtxt (set TextFileSource-use_pandas: True in config.yaml to use pandas instead)' % filename)
             self.res = np.genfromtxt(filename,
                              comments=comments,
                              delimiter=delimiter,
```

David-Baddeley, 2023-10-02:
no major objections - I think `logger.info` is probably sufficient - the default level is debug for PYME at the moment.

David-Baddeley, 2023-10-02:
see also #1429 for new warnings handling

csoeller, 2023-10-09:
Should be mostly done now with one PR already accepted and another one in the pipeline to reduce warnings where possible and turn print statements into log messages...

closed 2023-10-09 by csoeller