python-microscopy/python-microscopy #1426: [BUG] PYMEImage crosshairs x,y and I display not updating

barentine, 2023-09-18:
**Describe the bug**
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. open PYMEImage
2. advance to a frame
3. click to move the crosshairs
4. See the intensity of 0,0,0 (x, y, z/t) instead

David-Baddeley, 2023-09-18:
Platform, versions, data file type, and any error messages on the console?

I can't reproduce on Mac, Py3.8

barentine, 2023-09-18:
Interesting - thanks for checking, @David-Baddeley .
Windows, Py37, WX 4.0.4. Current repo head.

<details> <summary> Conda list </summary>

```
# packages in environment at C:\Users\aesb\Miniconda3\envs\pymedev:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py37_0    anaconda
anyio                     3.5.0            py37haa95532_0    anaconda
appdirs                   1.4.4                    pypi_0    pypi
argon2-cffi               21.3.0             pyhd3eb1b0_0    anaconda
argon2-cffi-bindings      21.2.0           py37h2bbff1b_0    anaconda
atomicwrites              1.4.0                      py_0    anaconda
attrs                     22.1.0           py37haa95532_0    anaconda
babel                     2.11.0           py37haa95532_0    anaconda
backcall                  0.2.0              pyhd3eb1b0_0    anaconda
beautifulsoup4            4.11.1           py37haa95532_0    anaconda
bic-gmm                   23.6.21                   dev_0    <develop>
blas                      1.0                         mkl    anaconda
bleach                    4.1.0              pyhd3eb1b0_0    anaconda
blosc                     1.21.3               h6c2663c_0    anaconda
bottleneck                1.3.5            py37h080aedc_0    anaconda
brotli                    1.0.9                h2bbff1b_7    anaconda
brotli-bin                1.0.9                h2bbff1b_7    anaconda
brotlipy                  0.7.0           py37h2bbff1b_1003    anaconda
bzip2                     1.0.8                he774522_0    anaconda
ca-certificates           2023.01.10           haa95532_0    anaconda
cairo                     1.16.0               h1c23506_3    anaconda
cattrs                    23.1.0                   pypi_0    pypi
certifi                   2022.12.7        py37haa95532_0    anaconda
cffi                      1.15.1           py37h2bbff1b_3    anaconda
chardet                   4.0.0           py37haa95532_1003    anaconda
charset-normalizer        2.0.4              pyhd3eb1b0_0    anaconda
cheroot                   8.5.2            py37haa95532_1    anaconda
cherrypy                  18.6.0           py37haa95532_0    anaconda
cloudpickle               2.0.0              pyhd3eb1b0_0    anaconda
colorama                  0.4.6            py37haa95532_0    anaconda
conda                     23.1.0           py37haa95532_0    anaconda
conda-build               3.23.3           py37haa95532_0    anaconda
conda-package-handling    2.0.2            py37haa95532_0    anaconda
conda-package-streaming   0.7.0            py37haa95532_0    anaconda
cryptography              38.0.4           py37h21b164f_0    anaconda
cycler                    0.11.0             pyhd3eb1b0_0    anaconda
cython                    0.29.32          py37hd77b12b_0    anaconda
cytoolz                   0.12.0           py37h2bbff1b_0    anaconda
dask-core                 2021.10.0          pyhd3eb1b0_0    anaconda
debugpy                   1.5.1            py37hd77b12b_0    anaconda
decorator                 5.1.1              pyhd3eb1b0_0    anaconda
defusedxml                0.7.1              pyhd3eb1b0_0    anaconda
django                    2.1                      py37_0    anaconda
docutils                  0.18.1           py37haa95532_3    anaconda
double-helix              23.5.4                    dev_0    <develop>
entrypoints               0.4              py37haa95532_0    anaconda
esbonio                   0.16.1                   pypi_0    pypi
exceptiongroup            1.1.1                    pypi_0    pypi
expat                     2.4.9                h6c2663c_0    anaconda
fftw                      3.3.9                h2bbff1b_1    anaconda
filelock                  3.9.0            py37haa95532_0    anaconda
flit-core                 3.6.0              pyhd3eb1b0_0    anaconda
fontconfig                2.14.1               hc0defaf_1    anaconda
fonttools                 4.25.0             pyhd3eb1b0_0    anaconda
freetype                  2.12.1               ha860e81_0    anaconda
fsspec                    2022.11.0        py37haa95532_0    anaconda
future                    0.18.2                   py37_1    anaconda
glib                      2.69.1               h5dc1a3c_2    anaconda
glob2                     0.7                pyhd3eb1b0_0    anaconda
gst-plugins-base          1.18.5               h9e645db_0    anaconda
gstreamer                 1.18.5               hd78058f_0    anaconda
[... 212 lines cut ...]
zstandard                 0.18.0           py37h2bbff1b_0    anaconda
zstd                      1.5.2                h19a0ad4_0    anaconda

```

</details>

I see now it does actually update the first time I click, but not thereafter, and there is a traceback:
```

ERROR:PYME.DSView.displayOptions:Error in OnChange
Traceback (most recent call last):
  File "c:\userfiles\andrew\code\python-microscopy\PYME\DSView\displayOptions.py", line 332, in OnChange
    fcn()
  File "c:\userfiles\andrew\code\python-microscopy\PYME\DSView\dsviewer.py", line 282, in update
    statusText += '    I: (%s)' % ', '.join(['%3.3f' % self.do.ds[self.do.xp, self.do.xp, self.do.zp, self.do.tp, c] for c in range(self.do.ds.shape[4])])
  File "c:\userfiles\andrew\code\python-microscopy\PYME\DSView\dsviewer.py", line 282, in <listcomp>
    statusText += '    I: (%s)' % ', '.join(['%3.3f' % self.do.ds[self.do.xp, self.do.xp, self.do.zp, self.do.tp, c] for c in range(self.do.ds.shape[4])])
  File "c:\userfiles\andrew\code\python-microscopy\PYME\IO\DataSources\BaseDataSource.py", line 348, in __getitem__
    return out.squeeze(axis=tuple(to_squeeze))
ValueError: cannot select an axis to squeeze out which has size not equal to one

```

Note that I opened an h5 file

David-Baddeley, 2023-09-19:
Confirmed with .h5 on mac ...

David-Baddeley, 2023-09-19:
And now fixed in #1427

David-Baddeley, 2023-09-19:
Specific bug was a bit nasty ... would only trigger on non-square images when you clicked outside the square region and then disabled updates for subsequent clicks.

closed 2023-09-19 by David-Baddeley

barentine, 2023-09-19:
Yikes - glad you found it. Thanks!