python-microscopy/python-microscopy #1428: [ENHANCEMENT] PYMEVis Snapshot option with bound-awareness

barentine, 2023-10-02:
**Is your feature request related to a problem? Please describe.**
Unless I'm mistaken, we don't have a great method of specifying the ROI of a generic snapshot, which can be a problem if someone is trying to e.g. create one image in ImageJ of an ROI and then generate another figure in PYMEVis (e.g. of labeling results, or a multi-layer view containing surfaces and localizations, etc.) which exactly matches the same ROI.

**Describe the solution you'd like**
I would like the View > Snapshot to have a combobox pop-up asking whether to grab the full view, or the `View Clipping` clipped view.

**Describe alternatives you've considered**
- A density-mapping 'view' recipe module so one could select a color-on key and other view parameters before passing to e.g. RGBImageOutput
- even the ability to draw a bounding box (volume) as we once had would be a helpful start 

**Additional context**
Trying to help a collaborator generate a figure for publication

David-Baddeley, 2023-10-02:
Matching a snapshot (rather than a rendered image) to an ImageJ snapshot is always going to be a bit tricky due to the arbitrary zoom of the OpenGL views. It becomes almost impossible once you add rotation. Before we had surfaces it was pretty easy to say that you should just use the generate functions instead, which give consistent pixel sizes and generally better quantitative accuracy (the OpenGL views take lots of shortcuts meaning that things like transparency are only approximate, so, e.g. the Gaussian rendering option is generally much better quality than the point-sprites equivalent). With 
surfaces, you do have a bit more of a legitimate use case for trying to match up a snapshot with a rendered image, but still only really in a strictly xy projection (which is often not what you want when looking at surfaces).

As to matching/specifying the ROI, clearing the clipping and then selecting `View->Fit` or `View->Fit ROI` before taking the snapshot should still work. I generally prefer the idea of clearing clipping manually (this could be a bit easier - potentially with the option to remember the last view and toggle) to having a combo-box when taking a snapshot - the latter adds lots of complexity. 

View clipping aside, the problem with `View->Fit` etc ... is that it doesn't get the aspect ratio right - you'll be left with a bit of dark space (typically at the bottom) to account for the different aspect ratios of the displayed view and the data. I could see an argument for adding an option to, e.g. crop to data bounds to the snapshot, although there are lots of complexities in getting this working properly and you'd have to, e.g., disable the option if not in a top-down view. Possibly an easier option would be to add a "Fit view window to data" option to the view which actually changed the aspect ratio of the GL display to match that of the data.

Bounding box is still there (`Extras->Scale box')

David-Baddeley, 2023-10-02:
It actually looks like most of the logic for doing what I *think* you want (i.e. matching viewport to data bounds, irrespective of view clipping) is already there. Let me know if this snippet (runs in PYMEVis console) does what you want, and we can put it behind a menu function along with a warning if you're not in a 2D view and/or adapt the current snapshot function.

```python
import PIL
s = glCanvas.getIm(pixel_size=30., image_bounds=pipeline.imageBounds).transpose(1,0,2)[::-1, :,:]
PIL.Image.fromarray(s).save('snap001.png')
```

barentine, 2023-10-02:
This is great,  @David-Baddeley , thank you! 

Yes, your snippet is what I'm after. Would be nice to have this behind a menu function. Note that if we do, we'll want to make a fresh ImageBounds instance since `pipeline.imageBounds` is not dynamically set.

```
>>> pipeline.imageBounds
ImageBounds(x0=4445.0, y0=457.0, x1=42705.0, y1=34512.0, z0=0.0, z1 = 0.0) instance at 0x1859C974DC8
>>> pipeline.filterKeys
{'x': (17055.370885656208, 20653.70849912825), 'y': (20124.0, 23429.97268237744)}
>>> from PYME.IO.image import ImageBounds
>>> ImageBounds.estimateFromSource(pipeline)
ImageBounds(x0=17057.0, y0=20126.0, x1=20653.0, y1=23429.0, z0=0.0, z1 = 0.0) instance at 0x18599D86448
>>> s = glCanvas.getIm(pixel_size=30., image_bounds=ImageBounds.estimateFromSource(pipeline)).transpose(1,0,2)[::-1, :,:]
30.0 30.0
Viewport:  (119, 110)
>>> PIL.Image.fromarray(s).save('snap003_test231002.png')
```

David-Baddeley, 2023-10-02:
OK - my thought in using the original image bounds (not those for the current ROI) was that this was more likely to be a good match to whatever you had opened in ImageJ. Of course, if the workflow is open localisations, set ROI, generate a density image, open that in ImageJ, then you want the bounds that were used for the generating the density image (which is likely to be the current ROI unless you then change it). It might make sense to have 3 options (match current view, match original data, match ROI).

David-Baddeley, 2023-10-02:
If we really wanted to be fancy, we could have a dropdown which also let you match the bounds of any image you had open in PYMEVis (caveat here is that it could only have been processed in PYME, otherwise the bounds metadata is likely to have gone missing / been corrupted).