python-microscopy/python-microscopy #1433: [BUG] (mostly harmless) error when cancelling .csv and .mat file opening in PYMEVis

csoeller, 2023-10-09:
**Describe the bug**
An error is raised when the tabular dialogs for `.csv` and `.mat` files are canceled

Note: error is largely inconsequential but should ideally be avoided. Fix is quite simple, PR to follow...

**To Reproduce**
Steps to reproduce the behavior:
1. Ty opening `.cvs` file in PYMEVis
2. Cancel tabular dialogs that show field mapping 
4. See error

**Error Message**

```
Traceback (most recent call last):

  File "/Users/csoe002/miniconda3/envs/pyme38_2/lib/python3.8/site-packages/wx/core.py", line 2346, in Notify

    self.notify()

  File "/Users/csoe002/miniconda3/envs/pyme38_2/lib/python3.8/site-packages/wx/core.py", line 3552, in Notify

    self.result = self.callable(*self.args, **self.kwargs)

  File "/Users/csoe002/Documents/src/PYMEsrc/pyme38/PYME/LMVis/visCore.py", line 800, in OpenFile

    self.pipeline.OpenFile(filename, **args)

TypeError: OpenFile() argument after ** must be a mapping, not NoneType
```

**Expected behavior**
No error.

David-Baddeley, 2023-11-28:
@csoeller - can we close this now #1434 has been merged?

csoeller, 2023-11-29:
fixed with #1434

closed 2023-11-29 by csoeller