python-microscopy/python-microscopy #1444: [SUPPORT] camera splitter related updates

csoeller, 2023-11-15:
We are eventually picking up on updating our splitter approach and use the latest code for acquisition with a CMOS camera (Andor Zyla/Sona). A number of questions have arisen. Finally, there is also an enhancement request (with proposed implementation).

For most of these issues I have some fixes in an experimental branch in my PYME working fork (branch `splitter-border-test`). But these are probably not yet quite ready for a PR but rather need a little discussion first, hence the support issue.

Nevertheless, I would like to address these all in one PR eventually, so hope to discuss here first and then raise as PR when I have something suitable.

### Issues brief overview

1. Getting unmix to work properly with a `left_right` axis CMOS setup. (workaround and proposed enhancement implemented)
2. In looking into the issue with unmix some questions about the deprecated `getROIX1` etc came up. While these calls obviously should be avoided in current code, the deprecated implementation in the `PYME.Acquire.Hardware.Camera` baseclass seems to have some issues. (workaround implemented)
3. Working with split acquisition data the `PYME.LMVis.colourPanel` is a little buggy with regard to some `wx` align options that trigger `assert`s and thus interfere with the `Add`/`Delete` popup menus not showing up. (workaround implemented)
4. Also with the `PYME.LMVis.colourPanel`, on windows only there is an issue with right clicking onto an existing `lFluorSpecies` entry that seems to come down to the event bound with the `wx.EVT_COMMAND_RIGHT_CLICK` event type which we have not resolved yet. (I'll explain further below)

The sections below expand on the specific issues.

### 1. Unmix with `left_right` axis splitter setup

Currently, unmix with `left_right` axis and shiftfield set is broken. Part of the issue seems to reside in the use of the ROI argument (in `PYME.Acquire.Hardware.splitter.py`) as used in the call

```
self.unmixer.Unmix(dsa, self.mixMatrix, self.offset, ROI=[self.scope.cam.GetROIX1(),self.scope.cam.GetROIY1(),self.scope.cam.GetROIX2(), self.scope.cam.GetROIY2()])
```

This brought me to the `GetROIXXX` implementations.

### 2. (Deprecated) Implementation of `GetROIX1/X2/Y1/Y2` partially buggy?

The description about these says in the `Camera.py` base class:

*Due to the fact that they map directly to API functions related to the Andor cameras, GetROIX1 and GetROIY1, return indices which are 1-based and GetROIX2 and GetROIY2 match x2 and y2 as provided to SetROI.*

The implementation (even if deprecated) however has:

    def GetROIX1(self):
        ...
        return self.GetROI()[0] + 1

    def GetROIX2(self):
        ...
        return self.GetROI()[2] + 1

    def GetROIY1(self):
        ...
        return self.GetROI()[1]

    def GetROIY2(self):
        ...
        return self.GetROI()[3]

Am I right that `GetROIX2` and `GetROIY1` are incorrectly implemented? They have basically been confused with each other?

And I checked, this leads to weird ROI values being passed to the `self.unmixer.Unmix` call which then promptly bombs.

### Proposed fix

1. Fix the deprecated `GetROIXY` calls and replace them with direct `GetROI` calls with clear logic where still present.
2. Overhaul the ROI logic for unmixing from the ground. I have done this already in a branch for the `left_right` case. The `up_down` axis case currently seems hardcoded for the Andor Ixon anyway.

**Note**: I realise that the unmix function with shiftfield set is mostly for visual checking that the corrections work as intended but still would like to get this working again.

### 3. Some minor `PYME.LMVis.colourPanel` issues

With split data the Colour panel comes into play in `PYMEVisualise`. The issue relates to editing/adding fluorophores using the listctrl widgets.

[image]

With current wx (mac: `4.2.0 osx-cocoa (phoenix) wxWidgets 3.2.0`/win `4.0.4 msw (phoenix) wxWidgets 3.0.5`) a few wx align options in `class specAddDialog(wx.Dialog)` (in `PYME.LMVis.colourPanel.py`) give rise to `assert` errors. This is easily fixed and makes everything work fully on mac again.

On win, editing existing fluorophores is broken, as right clicking on an existing entry **only** raises an `wx.EVT_COMMAND_RIGHT_CLICK` event bound via

        # for wxMSW
        self.lFluorSpecies.Bind(wx.EVT_COMMAND_RIGHT_CLICK, self.OnSpecListRightClick)

and does not raise

        # for wxGTK
        self.lFluorSpecies.Bind(wx.EVT_RIGHT_UP, self.OnSpecListRightClick)

The issue arises as the event that the `self.OnSpecListRightClick` sees is a `<wx._core.CommandEvent object>` and **NOT** a  `<wx._core.MouseEvent object>` (which the `wx.EVT_RIGHT_UP` correctly raises). The `wx._core.CommandEvent` lacks a `GetX` method and thus bombs. Net result, no context menus come up for editing/adding.

Strangely, when right-clicking elsewhere in the `lFluorSpecies` listCtrl, a `wx.EVT_RIGHT_UP` event **IS** raised *even on windows* and the `Add` context menu comes up. But no such luck when clicking on an existing fluorophore species, thus currently no editing. How to fix?

## 4. Proposed enhancement for `left_right` axis splitter setup without flipping

The proposed splitter setting enhancement concerns constrained ROIs with the splitter and is useful for the large chip sizes of the CMOS cams that we now routinely use and with a splitter that does not flip. Our illumination spot is smaller and thus we currently use about double the size ROI that we really need.

The idea is to use a symmetric border with a preset number of pixels on both sides of the central area that is used for imaging. One typically selects this in the ini file when the splitter is added for the local config.

The geometry looks as below:
[image]

It is quite easy to implement this with a simple change in `acquiremainframe.py` when ROIs are set:

	if self.scope.cam.splitting.lower() == 'up_down': 
        y1 = min(y1, self.scope.cam.GetCCDHeight() - y2) 
        y2 = max(y2, self.scope.cam.GetCCDHeight() - y1) 
     
        if not self.scope.cam.splitterFlip: 
           if 'splitterBorder' in dir(self.scope.cam): 
               y1 = self.scope.cam.splitterBorder 
               y2 = self.scope.cam.GetCCDHeight() - self.scope.cam.splitterBorder 
           else: 
               y1 = 0 
               y2 = self.scope.cam.GetCCDHeight()

The `splitterBorder` parameter is set in the splitter part of the init file, e.g.

```
@init_gui('splitter')
def ini_splitter(MainFrame,scope):
    from PYME.Acquire.Hardware import splitter
    constrainwidth = 550 # we constrain the width of the ROI region to 550 pixels; maximum is 1024 for the whole width
    chipwidth = scope.cameras['Zyla'].GetCCDWidth()
    scope.splt = splitter.Splitter(MainFrame, scope, scope.cameras['Zyla'],
                                   flipChan = 0, dichroic = 'BS50-50',
                                   transLocOnCamera = 'Left', flip=False,
                                   dir='left_right', constrain=False,
                                   border=int(int(chipwidth/2)-constrainwidth))
```

Turns out that in practice one seems to care more about the actual width that is available. So perhaps an alternative keyword `constrainWidth` or similar could be provided that does the math done in the call above.

We tested this already with shiftfield and detection etc, it seems completely fine. The change for the case `dir=top_down` is essentially the same, so the approach is not limited to `left_right`.

David-Baddeley, 2023-11-15:
It looks like the splitting code needs a **major** cleanup, with lots of duplication. At the moment, I'd regard the live splitting, and even the unsplit function in PYMEImage as being pretty broken - at least on non-trivial (up-down) splits. The splitter fitting which has seen a lot more recent attention, however, should be fairly robust. Ideally everything should be refactored to use `PYME.localization.splitting.get_splitter_rois()` and remove duplication with calculating ROIs etc ... .

## Issue 1

I think some of these (well issue 1 in any case) get fixed by specifying the `rois` parameter when creating the splitter object, e.g.:

```python
#splitter
@init_gui('Splitter')
def splitter(MainFrame, scope):
    from PYME.Acquire.Hardware import splitter
    splt2 = splitter.Splitter(MainFrame, scope, scope.cameras['sCMOS'], flipChan = 1, dichroic = 'T686lpxr' ,
                              transLocOnCamera = 'Right', flip=False, dir='left_right', constrain=False, cam_name='sCMOS', rois=[[70,0,800,2046], [880,0,800,2046]])

```

It was my intent to switch all the code to using this as it's much more flexible. If you pass the ROIs, it only uses the `dir` parameter if flipping the image to decide which axis to flip. I really need to add some deprecation warnings if the `roi` parameter is not provided.

## Issue 2

The preferred solution would be to remove all calls to `GetROIX1` etc .... I'd probably go a step further and base everything on `PYME.localization.splitting.get_splitter_rois()`. This would require some re-work of `PYME.Analysis.splitting` (started, but incomplete), and `PYME.IO.DataSources.UnsplitDataSource`.

## Constrain ROIs

This would likely need a fair bit of work to fix, and needs a major refactor. Ultimately I feel uncomfortable with having the splitter ROI constraint logic sitting in acquiremainframe. The good news is that it's no longer required for analysis (the logic in `PYME.localization.splitting.get_splitter_rois()` just takes the largest valid common ROI in the data that you've acquired.

## GUI stuff

You seem to be on the right track - nothing overly contentious here.

David-Baddeley, 2023-11-22:
I've had a quick go at some refactoring in #1445 - as yet incomplete, but should be helpful in indicating direction

csoeller, 2023-11-29:
Thanks @David-Baddeley, #1445 is helpful. We have started looking into some of the remaining issues and practical testing is under way in our lab. Even with the start on refactoring there are still disjunct places where the splitter channel ROI extraction logic is re-implemented in various ways so that overall behaviour can be inconsistent, sometimes unexpected and also potentially buggy. We will work on identifying those places and see if we can come up with a common place where the logic is (1) described how it should operate and (2) contains the actual implementation.

This looks like a slightly medium term project and we will probably come back with more details in the new year, possibly a little earlier...

csoeller, 2024-03-01:
Should be all done with #1445.

closed 2024-03-01 by csoeller