python-microscopy/python-microscopy #1465: [BUG] Bakeshop does not currently allow for localizations to be read from csv files

abalaji0827, 2024-02-08:
**Describe the bug**
When batch running a recipe with bakeshop, I am unable to use csv files for my input.

**To Reproduce**
Steps to reproduce the behavior:
1. open bakeshop
2. load recipe
3. load csv file to run recipe on
4. See error

**Expected behavior**
I expect the recipe to run as it usually does for hdf input

barentine, 2024-02-08:
I think @abalaji0827 is hitting an error upstream of table names. Looks like we haven't allowed csv loading in bakeshop, if I'm not wrong about loading in [PYME.recipes.recipe.loadInput](https://github.com/python-microscopy/python-microscopy/blob/e011b873d24ed1eb87d16505e8b81c12ca1c5aaa/PYME/recipes/recipe.py#L726) . This of course deviates from the PYMEVis loading [in pipeline.py](https://github.com/python-microscopy/python-microscopy/blob/e011b873d24ed1eb87d16505e8b81c12ca1c5aaa/PYME/LMVis/pipeline.py#L633).

David-Baddeley, 2024-02-08:
I'm sure we had it in there at one point - I wonder if this is an inadvertent regression. In any case, it should be fairly easy to fix.

David-Baddeley, 2024-02-08:
It looks like you are right - I was sure .csv loading was in there - maybe it regressed.

We should be able to copy most of the logic from `pipeline._ds_from_file(...)` (reproduced below) to `recipe.loadInput`

```python
else: #assume it's a delimited (tab or csv) text file
            # use provided `text_options` argument to Open(), or guess using csv_flavours
            text_options = kwargs.get('text_options', None)
            if text_options is None:
                # we didn't get any info about how to interpret the text file, guess
                logger.info('No text file format info provided, guessing ...')
                from PYME.IO import csv_flavours
                text_options = csv_flavours.guess_text_options(filename)

            ds = tabular.TextfileSource(filename, **text_options)
```
preceded by a `with unifiedIO.local_or_temp_filename(filename) as fn:`
and with the following caveats:

- default for recipes is to assume image, so we need to explicitly check  for .csv, .txt, **and** any additional extensions defined in `csv_flavours` rather than using a catch-all `else` assumption
- we might need to always guess the flavour 
- we should add the recipe default metadata to the datasource