python-microscopy/python-microscopy #1466: [BUG] Can't crop image, then run localization analysis on it

barentine, 2024-02-09:
**Describe the bug**

Cropping an image before localization analysis hits a bug
```
Traceback (most recent call last):
  File "c:\userfiles\davis_perez\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 269, in <lambda>
    self.bGo.Bind(wx.EVT_BUTTON, lambda e : self.lmanal.OnGo(e))
  File "c:\userfiles\davis_perez\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 662, in OnGo
    self.analysisController.pushImagesCluster(self.image)
  File "c:\userfiles\davis_perez\code\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 343, in pushImagesCluster
    self.resultsMdh['DataFileID'] = fileID.genDataSourceID(image.dataSource)
AttributeError: 'ImageStack' object has no attribute 'dataSource'
```

Note that if you try saving the cropped image first, you hit an issue which appears to be from saving the crop slices in metadata as slice objects:
```
DEBUG:PYME.cluster.rules:DataSource.__class__: <class 'PYME.IO.DataSources.HDFDataSource.DataSource'>
DEBUG:PYME.cluster.rules:local_queues: ['PYMERuleServer:WEM12A - PID:14352']
DEBUG:root:resultsURI: http://127.0.0.1:54947/__aggregate_h5r//stage_testing/2024_02_08_rebuild_stability_zstacks/analysis/CT_stability_servo_off_crop.h5r
Traceback (most recent call last):
  File "c:\userfiles\davis_perez\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 511, in default
    return obj.to_JSON()
AttributeError: 'slice' object has no attribute 'to_JSON'
```

**To Reproduce**
Steps to reproduce the behavior:
1. run PYMEClusterOfOne
2. load an h5 image file
3. select a region and crop
4. activate LMAnalyzer plugin in cropped PYMEImage window
5. click go
6. see error

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] Win10
 - Python version: believe 3.7, would have to check with labmate who's computer we were using
 - numpy version:
 - wxpython version [if a UI bug]:

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2024-02-09:
Localisation analysis needs the file to be saved (won't work with an in-memory data - i.e. the results of cropping, clipping, etc ...) as the file needs to be able to be opened by the analysis processes. The best way to handle this is probably to make sure that the error message is informative. 

The one potential wrinkle here is that, as it stands, just saving is not going to be enough - you need to save and re-open. We could look at making this un-necessary (i.e. just need to save) although it's hard to know how important that would be / if there was enough interest to make it worthwhile.

barentine, 2024-02-09:
Excellent point, David. I wasn't thinking about how the data actually gets distributed.

One thing to note is that saving a cropped imagestack to disk and reloading for loc analysis hits the second error, as the slice objects stored in metadata aren't readily json serialized.

barentine, 2024-02-09:
Looks like cropping metadata (which is never used in main PYME repo, only assigned) was a tuple until #1407 . 
Looks like slice objects can't be subclassed, so adding a to_JSON method is out.
OK to modify `PYME.IO.DataSources.CropDataSource._slice` to wrap in a slice object but return a tuple (slice.start, slice.stop, slice.step)?

David-Baddeley, 2024-02-10:
We should double check that `_slice()` is only used in metadata generation, otherwise generate the indices when assigned to the metadata. We might want to use `slice.indices(...)` rather than `slice.start`, `slice.stop` and `slice.step` as it's possible for `slice.stop` to be wrong (larger than data size) and still work for slicing, but not for working out what the cropped region was.