python-microscopy/python-microscopy #1487: [BUG] bug in quad tree construction during drift correction

csoeller, 2024-03-16:
**Describe the bug**
A record insertion error is triggered in `AutoFit` during drift correction with `PYMEnf`.

**To Reproduce**
Steps to reproduce the behavior:
1. Try `AutoFit` with `Model Based drift correction`
4. See error traceback further below

**Suspected cause of bug**
I do not know the code for quad tree construction in any detail, but the following patch makes some sense to me, and, if correct, I'd be happy to submit as PR. With this patch the drift correction worked (as well as expected).

```diff
diff --git a/PYME/Analysis/points/QuadTree/pointQTF.py b/PYME/Analysis/points/QuadTree/pointQTF.py
index 7f3da50b3..66be3e2a7 100644
--- a/PYME/Analysis/points/QuadTree/pointQTF.py
+++ b/PYME/Analysis/points/QuadTree/pointQTF.py
@@ -100,7 +100,7 @@ class qtLeafNode(qtNode):
             newBranch = qtBranchNode(self.x0,self.x1, self.y0, self.y1, self.depth)
             #for r in self.records: #copy records
             newBranch.insertRecords(self.records[:self.numRecords, :])
-            newBranch.insert(recs)
+            newBranch.insertRecords(recs)
             return newBranch
```

**Error traceback**
```
Traceback (most recent call last):

  File "/Users/csoe002/miniconda3/envs/test-pyme-3.10-conda_1/lib/python3.10/site-packages/traitsui/wx/ui_base.py", line 47, in perform

    self.ui.do_undoable(handler.perform, self.ui.info, self.action, event)

  File "/Users/csoe002/miniconda3/envs/test-pyme-3.10-conda_1/lib/python3.10/site-packages/traitsui/ui.py", line 658, in do_undoable

    action(*args, **kw)

  File "/Users/csoe002/miniconda3/envs/test-pyme-3.10-conda_1/lib/python3.10/site-packages/traitsui/handler.py", line 224, in perform

    method(info)

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/pymenf-p310_1/PYMEnf/recipes/pymenf.py", line 191, in on_autofit

    params = parent.autofit()

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/pymenf-p310_1/PYMEnf/recipes/pymenf.py", line 89, in autofit

    xp, yp, ip = ROIIdentify.GetConstantPointROIs(self._ds, 10000)

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/pymenf-p310_1/PYMEnf/DriftCorrection/ROIIdentify.py", line 13, in GetConstantPointROIs

    qt = pointQTF.createQT(pipeline['x_raw'], pipeline['y_raw'],np.arange(len(pipeline['x'])))

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/python-microscopy-python-310-compat/PYME/Analysis/points/QuadTree/pointQTF.py", line 252, in createQT

    qt.insert(np.array((xi,yi, ti)))

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/python-microscopy-python-310-compat/PYME/Analysis/points/QuadTree/pointQTF.py", line 223, in insert

    self.root = self.root.insert(rec)

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/python-microscopy-python-310-compat/PYME/Analysis/points/QuadTree/pointQTF.py", line 146, in insert

    self.NW = self.NW.insert(rec)

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/python-microscopy-python-310-compat/PYME/Analysis/points/QuadTree/pointQTF.py", line 151, in insert

    self.NE = self.NE.insert(rec)

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/python-microscopy-python-310-compat/PYME/Analysis/points/QuadTree/pointQTF.py", line 146, in insert

    self.NW = self.NW.insert(rec)

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/python-microscopy-python-310-compat/PYME/Analysis/points/QuadTree/pointQTF.py", line 89, in insert

    newBranch.insertRecords(self.records[:self.numRecords, :])

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/python-microscopy-python-310-compat/PYME/Analysis/points/QuadTree/pointQTF.py", line 162, in insertRecords

    self.NE.insertRecords(recs[(W<1)*N, :])

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/python-microscopy-python-310-compat/PYME/Analysis/points/QuadTree/pointQTF.py", line 103, in insertRecords

    newBranch.insert(recs)

  File "/Users/csoe002/Documents/src/PYMEsrc/PYME-test-env/build-test-py3.10-conda_1/python-microscopy-python-310-compat/PYME/Analysis/points/QuadTree/pointQTF.py", line 144, in insert

    if rec[0] < self.xc:

ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - latest PYME from github on macOS

David-Baddeley, 2024-03-18:
That does look like it might make sense ... funny that it's only showing up now - is it on all data sets or just one?

csoeller, 2024-03-18:
I recently was working with some large data sets (150-220MB `h5r`s) with quite high (possibly too high) event density. Those data sets consistently hit the "bug". This may be consistent with this tree division only rarely exercised on "normal" data sets.

csoeller, 2024-04-09:
fixed with merged PR #1488

closed 2024-04-09 by csoeller