python-microscopy/python-microscopy #1490: Spherical Harmonics Fit Error

loekloek, 2024-03-21:
**What are you trying to do?**
I am trying to perform a Spherical Harmonic Fit to fit a surface to my (quite spherical) data.

**Where are you stuck?**
When running the 'Analysis/Surface Fitting / Fit Spherical Harmonic Shell I encounter the following error:
I tried to manually change self.modes to an np.array, but that did not work. Do you have an idea if I am doing something wrong? Let me know if you need more information.

All the help is appreciated.

 Error whilst running Analysis>Surface Fitting>Spherical Harmonic Fitting>Fit Spherical Harmonic Shell
==============================================
python-microscopy=23.05.17.post2[conda]
python=3.7.16, platform=win32
numpy=1.21.5, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "C:\Users\Loek\python-microscopy\lib\site-packages\PYME\recipes\recipe.py", line 237, in execute
    m.execute(self.namespace)
  File "C:\Users\Loek\python-microscopy\lib\site-packages\PYME\recipes\surface_fitting.py", line 495, in execute
    shell.table_representation.addColumn('median_residual', np.array([np.median(separations),]))
  File "C:\Users\Loek\python-microscopy\lib\site-packages\PYME\Analysis\points\spherical_harmonics.py", line 314, in table_representation
    self._update_table_repr()
  File "C:\Users\Loek\python-microscopy\lib\site-packages\PYME\Analysis\points\spherical_harmonics.py", line 331, in _update_table_repr
    modes = self.modes.astype('i4'),
AttributeError: 'list' object has no attribute 'astype'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\Loek\python-microscopy\lib\site-packages\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "C:\Users\Loek\python-microscopy\lib\site-packages\PYME\LMVis\Extras\spherical_harmonics.py", line 34, in OnCalcHarmonicRepresentation
    recipe.add_modules_and_execute([shell_maker,])
  File "C:\Users\Loek\python-microscopy\lib\site-packages\PYME\recipes\recipe.py", line 512, in add_modules_and_execute
    self.execute()
  File "C:\Users\Loek\python-microscopy\lib\site-packages\PYME\recipes\recipe.py", line 282, in execute
    raise RecipeExecutionError('Recipe execution failed in module %s' % self._failing_module._module_name, self ) from e
PYME.recipes.recipe.RecipeExecutionError: Recipe execution failed in module surface_fitting.SphericalHarmonicShell

David-Baddeley, 2024-03-21:
It definitely looks like it should be an array, and looking at the commit history it looks like I'm to blame for the offending change which triggered this. 

I suspect it could be fixed by wrapping either the setting of `self.modes` (occurs in a couple of places in Analysis/points/spherical_harmonics.py) in a `np.array()` call. 

Alternatively (and probably safer as there is less potential for undesired side effects) might be to wrap the access in `_update_table_repr()` - i.e. make line 330 read
`modes = np.array(self.modes).astype('f4').

Whichever way it is done, `self.coefficients` will need the same treatment.

loekloek, 2024-04-02:
Thanks for the fast response! Wrapping modes and coefficients in _update_table_repr() indeed worked.