python-microscopy/python-microscopy #1494: [BUG] dependency for newish fit _factory does not get built?

csoeller, 2024-03-28:
**Describe the bug**

For a few months now I have seen the following which looks like a (largely inconsequential) bug (?) on all platforms where we have built recent PYMEs. Essentially, registering the fit factory `LatGaussFitlmR` fails due to a missing `PYME.localization.lev_gfit.levmar_gfit` dependency. The suspicion is that it does not get built due to something in a setup file not being quite right or similar, but we are not certain and have not investigated deeply.

The bug (if you can call it that) is not a big deal but it would be good to get rid of the "noise" in the logs/output when starting `PYMEAcquire`. If it can be fixed easily so much the better.

```
ERROR:PYME.localization.FitFactories:Error registering fit factory: LatGaussFitlmR, LatGaussFitlmR, PYME.localization.FitFactories
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYMEsrc/pyme38/PYME/localization/FitFactories/__init__.py", line 56, in _register_fit_factory
    fm = importlib.import_module(pkg + '.' + module_name)
  File "/Users/csoe002/miniconda3/envs/pyme38_2/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/csoe002/Documents/src/PYMEsrc/pyme38/PYME/localization/FitFactories/LatGaussFitlmR.py", line 33, in <module>
    from PYME.localization.lev_gfit.levmar_gfit import *
ModuleNotFoundError: No module named 'PYME.localization.lev_gfit.levmar_gfit'
```

**To Reproduce**
Steps to reproduce the behavior:
1. start `PymeAcquire`
2. See bug

**Platforms (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Windows 10, 11, macOS

David-Baddeley, 2024-03-28:
Had to laugh at the "newish" in the title - I think that fit factory dates back to my Postdoc. It should probably be deprecated and removed (was an attempt to squeeze a little bit extra speed out of localisations, but only got around 10% and needed to be built with gcc - i.e. didn't work with standard windows toolchains). Has some historical interest / potential future interest in that it's essentially an all c-code implementation of the fitting, but doesn't belong  where it is constantly trying to be loaded.

David-Baddeley, 2024-04-08:
Should be fixed with above linked commits

closed 2024-04-08 by David-Baddeley