python-microscopy/python-microscopy #1496: [SUPPORT] pipeline mdh, datasources mdh and saving filtered localisations

csoeller, 2024-04-09:
I recent looked into metadata generated when recipe modules are executed. The metadata gets nicley autogenerated from the various modules (via code triggered in the module's `execute` method).

Now when I save the current output of the pipeline from `PYMEVisualise` (`File>Save filtered localizations`) to `.hdf` the contents of `pipeline.mdh` gets saved with it (it seems to me). The issue to me is that `pipeline.mdh` only contains metadata created at pipeline generation and *not* the metadata of the currently selected datasource.

Several questions arise:

1. Should `pipeline.mdh` stay as is from generation and *not* reflect the currently selected datasource?
2. Shouldn't the saving with `File>Save filtered localizations` not include the metadata from the currently selected datasource (and possibly from `Output Filter` metadata as well)?

In general, what is a reasonable intended relationship between `pipeline.mdh` and the `mdh`'s from the datasources in the current pipeline in `PYMEVisualise`?

There are also GUI questions if `pipeline.mdh` were mutable as the Metadata tab would possibly go out of sync, so may need some careful thinking what reasonable behaviour would be.

As is, the datasource's metadata stays essentially hidden in `PYMEVisualise` unless custom plugins/shell interaction explicitly were to access it.

Any ideas appreciated.

csoeller, 2024-04-09:
Here a few quick examples with PYMEVisualise settings as in screenshot below

```
pipeline.mdh
<NestedClassMDHandler>:

MINFLUX.Is3D: True
MINFLUX.ExtraIteration: False
```
versus

```
pipeline.dataSources['colour_mapped'].mdh
<DictMDHandler>:

MINFLUX.Is3D: True
MINFLUX.ExtraIteration: False
Processing.AddPipelineDerivedVars.Foreshortening: 1.0
Processing.AddPipelineDerivedVars.Pixelsizenm: 1
Processing.ProcessColour.RatiosFromMetadata: True
Processing.ProcessColour.SpeciesDyes: {}
Processing.ProcessColour.SpeciesRatios: {}
```

[image]

David-Baddeley, 2024-04-09:
Thanks for raising this! Some of this is fairly easy:

- when saving results which have been processed by the pipeline, we should really save all the relevant metadata that was generated during recipe execution.
- I think it's reasonable (at least in the first instance) if the metadata tab just shows the metadata for the file we have opened without any subsequent processing.

Implementation needs a (little) bit more thought ...

- in many ways it would be cleaner and more logical to allow saving of the recipe datasources, rather than pipeline output, **except** that these do not have cropping and/or channel filters applied.
- simplest thing to do might be to just make `pipeline.mdh` an alias for `pipeline.selectedDataSource.mdh` (this should give rise to the same static metadata in the metadata panel as it will be equivalent when an image is loaded). It would also make things like setting the metadata for simulated events much cleaner (you simply set the data source metadata rather than doing silly injection stuff). I can't really see any disadvantage to the alias route at present.
- We should also consider an interface for saving individual recipe datasources.

csoeller, 2024-04-10:
Very cool, thanks for looking into it. Do we mind that the `Metadata` tab in `VisGUI` stays on the datasource mdh autoselected at startup when other datasources are selected?