python-microscopy/python-microscopy #1504: [ENHANCEMENT] Quiver rendering without vertices (e.g. for dipole orientation visualization)

barentine, 2024-05-02:
**Is your feature request related to a problem? Please describe.**
All of the heavy lifting has already been done (thanks @zacsimile and @David-Baddeley !), but the `QuiverRenderLayer` is typed for `triangle_mesh.TrianglesBase` only.
We are interested in plotting single-molecule orientation microscopy (SMOLM) data where we do not have a surface, but do have XYZ and 3D orientation vectors (mu_x, mu_y, mu_z) for each localization. 
We're happy to re-tool what's there a bit, but wanted to check in first to see a) whether I missed something and there is already a convenient way to visualize this in PYMEVis without modifying anything, or b) whether our approach would be the cleanest.

**Describe the solution you'd like**
- Subclass `QuiverRenderLayer` and rejig `update_from_datasource` so it accepts a normal localization datasource rather than TriangsBase.
- render each localization twice, with opposite orientations such that the point stays in the middle of the bar that is drawn

**Describe alternatives you've considered**
you having already written this somewhere ;)

**Additional context**
[active smolm work in our lab](https://doi.org/10.1016/j.optcom.2023.129589)
[style of visualizations we are aiming for (but interactively in 3D, of course!)](https://doi.org/10.1364/OPTICA.388157)

**Are you prepared to work on it?**
yes

David-Baddeley, 2024-05-20:
Sorry I've taken so long to get to this ... it came in at an awkward time and then got forgotten. 

- I don't think there is a solution already implemented.
- Subclassing `QuiverRenderLayer` seems reasonable, although we might want to rejig ever so slightly at some point to put the shared logic in a common base class instead.
- Instead of rendering each location twice, consider making another `Engine` which which subtracts from one end of the vector and adds to the other end instead. I think this would be a bit cleaner, and would be faster / less memory intensive. 

ie  duplicate the engine and replace

```python
    vec_buffer = np.empty((vertices.shape[0]+vecs.shape[0],3), dtype=vertices.dtype)
    vec_buffer[0::2,:] = vertices
    vec_buffer[1::2,:] = vertices
            
    vec_buffer[1::2,:] += layer.scaling*vecs
```  

with 

```python
    vec_buffer = np.empty((vertices.shape[0]+vecs.shape[0],3), dtype=vertices.dtype)
    vec_buffer[0::2,:] = vertices - layer.scaling*vecs
    vec_buffer[1::2,:] = vertices + layer.scaling*vecs
```

barentine, 2024-05-29:
Sounds great - thanks, @David-Baddeley !