python-microscopy/python-microscopy #1505: [BUG] wx._core.wxAssertionError when adding recipe modules

Yujin-Bao, 2024-05-06:
**Describe the bug**
Hi @David-Baddeley @barentine , sorry for the late notice, but I think this might come from the recent PRs. I used to use `wxpython` 4.0.4 but now I update it to 4.1.1 due to PR#1500 and PR#1503. Since then there is an error when I add some recipe modules. I realize such recipe modules generating this error have a common point that they need to read files (e.g. `PYME.recipes.traits.FileOrURI` and `PSFFile` in `processing.py`). And the full traceback is as below

```
Traceback (most recent call last):
  File "c:\users\yb262\python-microscopy\PYME\recipes\recipeGui.py", line 603, in OnAddModule
    if c.configure_traits(kind='modal'):
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traits\has_traits.py", line 2095, in configure_traits
    args,
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\toolkit.py", line 218, in view_application
    context, view, kind, handler, id, scrollable, args
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\view_application.py", line 101, in view_applic                                     ation
    args=args,
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\view.py", line 462, in ui
    ui.ui(parent, kind)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\ui.py", line 244, in ui
    self.rebuild(self, parent)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\toolkit.py", line 132, in ui_modal
    ui_modal.ui_modal(ui, parent)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\ui_modal.py", line 46, in ui_modal
    _ui_dialog(ui, parent, BaseDialog.MODAL)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\ui_modal.py", line 65, in _ui_dialog
    BaseDialog.display_ui(ui, parent, style)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\ui_base.py", line 71, in display_ui
    ui.owner.init(ui, parent, style)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\ui_modal.py", line 139, in init
    sw = panel(ui, window)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 274, in panel
    panel, content[0], ui
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 433, in fill_panel_for_grou                                     p
    panel, group, ui, suppress_label, is_dock_window, create_panel
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 589, in __init__
    self.add_items(content, panel, self.sizer)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\ui_panel.py", line 909, in add_items
    editor.prepare(item_panel)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\editor.py", line 236, in prepare
    self.init(parent)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\file_editor.py", line 96, in init
    sizer.Add(control, 1, wx.EXPAND | wx.ALIGN_CENTER)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL))" failed at ..\..\src\com                                     mon\sizer.cpp(2184) in wxBoxSizer::DoInsert(): Vertical alignment flags are ignored with wxEXPAND
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10
 - Python version: 3.7.11
 - numpy version: 1.21.5
 - wxpython version: 4.1.1

csoeller, 2024-05-06:
Hi @Yujin-Bao, I came across this issue a little while ago. There are various workarounds (or hacks) for a quick fix. I just up editing the offending line in `Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\file_editor.py` and removed the offending flag (I think it is the `wx.ALIGN_CENTER`). This then makes everything work again but is clearly not the proper fix.

The correct fix, I think, is to move to a more recent `traitsui` package. That, however, if I recall correctly, was not trivial since the more recent `traitsui` required another more recent module, I think it was `pyface` and that in turn had dropped some deprecated calls that were used by some other required PYME functionality. For that reason I had tied `traitsui==7.1.0 pyface==7.1.0`.

So the quick "hack" got me at least going again but we should look into the proper solution...

csoeller, 2024-05-06:
PS - I briefly built a new conda env with py3.10 and lifted the constraints on `traitsui` and `pyface`. This installed versions 8.0.0 of these two packages:

```
traitsui                  8.0.0              pyhd8ed1ab_0    conda-forge
pyface                    8.0.0              pyhd8ed1ab_0    conda-forge
```

With that one can use `FileOrURI` fine (I think traitsui >= 7.6.X has the alignment flag fix). I have not yet triggered the other issue I had at the time resulting from the more recent `traitsui` but only have a dim collection which functionality triggered it, it might have been something in contrib code but I am not sure.

More generally, the stricter alignment flag assertions I seem to recall coinciding with wx >= 4.1 and I had submitted PRs to fix any occurences in PYME (already merged).

David-Baddeley, 2024-05-06:
Yeah - I'm pretty sure I PR'ed a fix to traitsui for the file control some time back. I've been running traitsui=7.5 (with patch) for quite some time now without running into any issues. As I recall the main reason for pinning was that the conda packages for one version (probably around 7.2) dropped wx support altogether (potentially a build issue), but this has now been reinstated.

Yujin-Bao, 2024-05-07:
Thanks @csoeller @David-Baddeley ! I am currently using `traitsui==7.1.0 pyface==7.1.0`. I tried the "hac" and commented out the line 96 in `Installation\Miniconda\envs\pyme\lib\site-packages\traitsui\wx\file_editor.py` and it worked. I also tried updated the packages but the versions were not compatible. I think I need to get a newer python version in the future, too.

Yujin-Bao, 2024-05-08:
Something is still not quite right here - there is no icon to let me select the file. The screenshot is as follows. I was wondering if you had such issue?
[image]

csoeller, 2024-05-08:
You mention that you _commented out the line 96 in ..._. I think the idea is not to comment out the line but change it to

     sizer.Add(control, 1, wx.EXPAND)

i.e. only remove the alignment flag. Does that fix it?

Yujin-Bao, 2024-05-08:
Sorry for the misundersdanding. Yes, that fixed it. Thank you very much!

David-Baddeley, 2024-05-20:
I think we can close this subject to #1507

closed 2024-05-20 by David-Baddeley