python-microscopy/python-microscopy #1507: [ENHANCEMENT] Bump dependencies, fix packaging

David-Baddeley, 2024-05-20:
It's past time we bumped dependencies and package builds to newer versions, as we've run into a number of dependency issues recently.

Suggested default and supported configs below:

#### Default:

python=3.8
wx >= 4.1
traitsui >= 7.6

### Supported:

python=3.9, 3.10??

## Questions?

- Will we need to use conda-forge?  [probably]
- Should we be using mamba rather than conda?
- Can we build executable installers for apple silicon?
- Can we package bioformats?

csoeller, 2024-05-21:
I am running python=3.10 on production versions for a while now, still a few things to verify on an acquisition system (in the pipeline) but pretty flawless now for all kinds of analysis. There may still be rarely exercised code portions that are untested though but the only way to find out may be to ask people to use it...

zacsimile, 2024-05-21:
I'm on `miniforge` with the `mamba` solver for a while now. It's much faster.

We should be able to package bioformats but we could also consider moving away from using a Java bridge. Allen Inst has quite nice native support at this point in: https://github.com/AllenCellModeling/aicsimageio. We could use this to import. Alternatively, we could look at their packaging scheme for `bioformats_jar`.

barentine, 2024-05-30:
I appreciate the work you all have put into packing/dependencies. Certainly been an issue lately.

I'm taking miniforge/mamba and py3.10 for a spin using @csoeller 's [PYME-test-env.](https://github.com/csoeller/PYME-test-env).

[3.8 is EOL in 5 months](https://devguide.python.org/versions/). That of course doesn't mean we have to skip 3.8 as a default for now, but I'm happy to spend some time running into things on 3.10 so we can at least remove the question marks on supporting it.

barentine, 2024-10-30:
Looks like recent PR #1538  might kill compatibility with wxpython 4.2.1. I've been running 4.2.1, but can no longer open PYMEAcquire after merging main.

I'm hitting an error similar to [this one](https://discuss.wxpython.org/t/wxsizer-setcontainingwindow-issue-in-4-2-1/37020)

```
ERROR:PYME.ui.mytimer:Error in timer callback
Traceback (most recent call last):
  File "c:\userfiles\code\python-microscopy\PYME\ui\mytimer.py", line 48, in Notify
    a()
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\acquiremainframe.py", line 157, in _check_init_done
    self.doPostInit()
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\acquiremainframe.py", line 291, in doPostInit
    self.pan_spool = spool_panel.SpoolingPane(self, self.scope, acquisition_uis=self._aq_uis)
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\ui\spool_panel.py", line 574, in __init__
    self._init_ctrls()
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\ui\spool_panel.py", line 543, in _init_ctrls
    self.AddNewElement(self._analysis_pan())
  File "c:\userfiles\code\python-microscopy\PYME\Acquire\ui\spool_panel.py", line 379, in _analysis_pan
    pan.SetSizerAndFit(vsizer)
wx._core.wxAssertionError: C++ assertion "CheckExpectedParentIs(w, m_containingWindow)" failed at ..\..\src\common\sizer.cpp(887) in wxSizer::SetContainingWindow(): Windows managed by the sizer associated with the given window must have this window as parent, otherwise they will not be repositioned correctly.

Please use the window wxPanel@0000020CD37B9A40 ("panel", HWND=0000000000020752) with which this sizer is associated, as the parent when creating the window wxPanel@0000020CD37B6F40 ("panel", HWND=0000000000010788) managed by it.
```
rolling back to 4.1.1 fixed things for now

David-Baddeley, 2024-10-31:
@barentine - looks like a wx bug, which is frustrating. Curiously the offending code is essentially the same as what was there for all the other panels (and worked fine), which is going to make it hard to debug and work around. 

It might be useful to check to see how reproducible this is (i.e. does it also happen on mac, different windows machines, etc ...)

barentine, 2024-10-31:
Maybe I haven't pushed hard enough on it, but at first pass it seems not-so-reproducible. I had the wxpython 4.2.1 installed via pip on my desktop computer (python 3.10) and PYMEAcquire loads fine, meanwhile my instrument computer running wxpython 4.2.1 installed via conda (and python 3.9) hit the error above. Both machines are windows 10. 

For now, I've gotten around it by pipping in wxpython 4.1.2 on my py 3.9 instrument computer.

barentine, 2024-11-14:
hitting the same issue on another machine, win11, python 3.10, wxpython  4.2.2  py310h9e98ed7_1 conda-forge. Pip installing wxpython=4.2.2 doesn't help. 
Pipping in wxpython==4.1.1 doesn't immediately work, might need to install out of date msvc to get it work
```
File "C:\Users\Administrator\miniforge3\envs\pyme310\lib\site-packages\setuptools\_distutils\msvc9compiler.py", line 280, in query_vcvarsall

          raise DistutilsPlatformError("Unable to find vcvarsall.bat")

      distutils.errors.DistutilsPlatformError: Unable to find vcvarsall.bat
      Finished command: build_wx (0.381s)
      Finished command: build (0.381s)
      Command '"C:\Users\Administrator\miniforge3\envs\pyme310\python.exe" -u build.py build' failed with exit code 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for wxpython
  Running setup.py clean for wxpython
Failed to build wxpython
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (wxpython)
```

barentine, 2024-11-14:
ah, I think I've got the wxpython thing sorted. Seems to be a bug in our code. Will PR shortly and reference this issue, but wanted to comment here explicitly that I don't think we have a wxpython 4.2.2 compatibility issue - sorry for any confusion along that line