python-microscopy/python-microscopy #1516: [ENHANCEMENT] simple way to graphically delete a recipe module

barentine, 2024-06-06:
**Is your feature request related to a problem? Please describe.**
It's pretty unintuitive to have to delete recipe module text and update it. When folks see the GUI representation they want to work with that. 

**Describe the solution you'd like**
I think it coudl be nice to have an x or a trashcan someone can click on a module
[image]

**Describe alternatives you've considered**
when someone double clicks a module to edit it their could be a delete button, but that's a little harder to find maybe

David-Baddeley, 2024-06-06:
A GUI way of deleting modules has been on the TODO list ever since I wrote the recipe editor .... (and might be mentioned in a previous issue/roadmap). We are a little constrained in what we do because we use matplotlib to display the recipe, which makes it much harder to program interaction with the various bits.

My initial plan had been to make right-clicking on the module display a pop-up menu with delete as one of the options (like removing filter settings etc ...). This should be reasonably easy to implement, and has some GUI consistency, but has the disadvantage of being a bit hard to discover.

Adding a delete button to the edit box is a possibility (especially if we change the editor to also show module docs as mentioned in https://github.com/python-microscopy/python-microscopy/issues/1316#issuecomment-2153446375). It's also a bit gross though in terms of discoverability.

I don't like the idea of having a permanent cross / rubbish bin on the module, as it would be too easy to accidentally click (and hard to recover from) especially as clicking anything in the recipes relies on hit-testing a co-ordinate in matplotlib and is a bit fuzzy. We could mitigate this to some extent by having a confirmation dialog, but that would be annoying. 

What would be ideal is if mousing over the control made `edit` and `delete` options appear below the module. This still makes it two-step, as it requires a deliberate mouse movement from where you enter to where the `delete` option is, but would not be as annoying as having a confirmation  dialog. Unfortunately I think implementing this on top of the current matplotlib display would test ones sanity.

csoeller, 2024-07-08:
Ok to close with  #1519 merged?

David-Baddeley, 2024-07-09:
#1519 has the issue that the way of deleting modules is not easily discoverable, so feels a bit like a partial solution. Not sure if we should keep the issue open in the hopes of a better solution, or close and re-open.