python-microscopy/python-microscopy #1517: [ENHANCEMENT] un-hardcode some things in Corrections > Fiducials

barentine, 2024-06-06:
**Is your feature request related to a problem? Please describe.**
Corrections > Fiducials > Correct in `PYME\LMVis\Extras\fiducial.py` gives folks the opportunity to set their fiducial size, which is great, and then edit the FilterTable applied to fiducials, but they don't get a chance to edit the datasource name (hardcoded as `Fiducials`) nor the DCSCANClustering parameters. This was recently a problem for someone who wanted to drift correct their non-blinking localizations using the same data they had open, and the hack to use a blank MappingFilter or TableFilter to alias their localizations to Fiducials is kind of gross.

**Describe the solution you'd like**
It would be nice if we could .configure_traits(kind='modal') on more than just the filter, I suppose.
I believe we can do that for the whole recipe at once, but that might be a bit much.

David-Baddeley, 2024-06-06:
The standard approach for this use case is to call `Corrections->Fiducials->Load fiducial fits from 2nd file` and use the same file. This is actually (surprisingly) documented in the fiducial correction howto (although I'd need to check if this has already been converted to .rst and bundled with the main docs, or if it's still a standalone .doc / .pdf).

As the fiducial correction is a bit temperamental and hard to get right, I'm in favour of trying to keep the procedure fairly rigid and making people jump through (some minimal) hoops to depart from the procedure. Most notably, the defaults related to object detection and background subtraction when performing standard fits will make non-blinking fiducials disappear. This means that in most cases it is not appropriate to use the same set of localisations for reconstruction and fiducial correction. Whatever we do, we should make it obvious that this is not a recommended course of action.

David-Baddeley, 2024-06-06:
The docs are at:
https://github.com/python-microscopy/messy_pyme_docs/blob/master/Drift%20correction%20using%20fiducials%20in%20PYME.pdf

and haven't been converted.

I was also incorrect, in that the `Load fiducial fits from 2nd file` bit is not in there (I was certain it was, but maybe it's in a different version)