python-microscopy/python-microscopy #1521: [ENHANCEMENT] recipe input modules for some scenarios

csoeller, 2024-06-17:
**General issue:**

We find ourselves now semi-regularly with a use case where we would want to use an input module for an image type datasource in a PYMEVisualise recipe. This is typically an image that is used for masking localisations. While this can be achieved by `Open Raw/Prebeleach data` the main use case is that where you want to preserve the state of a PYMEVisualise session and quickly get back to that later - WITHOUT having to go through 3-5 GUI interactions to make it happen.

**Further background:**

Recipes seeme to be used in at least two quite distinct scenarios:

1. To run a type of computation on a number of datasets in kind of a loop (mostly with image data?), i.e. `bakeshop` et al.
2. To save the state of processing of a localisation file and allow reopening with that state loaded from a yaml.

The second case is the one we focus on mostly these days and here the ability to load a mask and associate it via recipe with the localisation dataset in question is the use case we frequently have.

**Current workaround solution:**

I made a fairly simple input module for something that can be loaded with the ImageStack class, as follows:

```python
from PYME.IO.image import ImageStack
        
@register_module('LoadMask')    
class LoadMask(ModuleBase):

    inputName = Input('FitResults') # we use this as a dummy input since we need at least one input
    outputImageStack = Output('image_mask')

    maskfile = FileOrURI('')

    def run(self,inputName):
        if self.maskfile != '':
            fp = Path(self.maskfile)
            if not fp.exists():
                warn("trying to load file '%s' that does not exist" % self.maskfile)
                return None
            mask = ImageStack(filename=self.maskfile)
            return mask
        else:
            return None # this will trigger an error as it tries to attach the mdh; may be better to have an empty container type
```

Some issues with that:

- it needs a dummy input (need at least one input per module with current code)
- it does not check extensions or similar
- it uses absolute paths (probably has to) and thus makes it more tricky to share with others

Nevertheless it serves a need reasoanbly well and serves as a proof-of-principle.

**Open questions:**

1. Is there space for a better way to implement this that acknowledges the two quite different use cases of recipes, see above?
2. WIth the idea of "preservation of state" is it conceivable to also save the layer arrangement with a recipe so that, for example, the layer layout could be restored with a recipe (or generalisation thereof)? This is interesting when you have several layers with specific datasource settings etc for a specific dataset.

**Caveats**

Apologies if I have overlooked existing functionality that already implements this.

David-Baddeley, 2024-06-17:
Hi Christian,

recipes were never intended for preserving state (even within PYMEVis), but rather preserving a set of operations that you might want to apply to multiple different files. This makes hard-coding data file names in a recipe somewhat antithetical to what they are designed for. The one exception here is for calibration files which are likely to be used for multiple datasets.

That said, I do appreciate the use case you have in mind, but it needs some thought to implement in a sensible way. The ideal solution would probably be somewhere along the following lines:

- allow you to add additional named data sources to the PYMEVis command line, much as you would to batchProcess  eg: `PYMEVis somefile.h5r -r some_recipe.yaml --mask=somefile_mask.tif` and have the data source load into the recipe namespace at the given name. This could re-use the `recipe.loadInput()` logic.
- Optionally (to make things a bit more user friendly), add a `Save Session` menu option to PYMEVis which essentially saves the recipe *and* the command line you would need to recapitulate it and load any additional data - maybe in a larger .yaml. That way, you could then re-load by running `PYMEVis saved_session.pvs` or similar. This would need a way of keeping track of the various additional channels you have opened (the open channel stuff could probably use a bit of an overhaul in any case). A "session" file would also be a good place to keep layer and view settings etc ...

I would really like to keep the recipes themselves as a description of a reproducible / transferrable processing pipeline rather than a description of state. 

Regardless of how we record sessions/state, there will be a problem of what to do with the various things that get generated independently of the recipe (either for legacy reasons, or to isolate things with lots of computational cost).

It's a bit moot, as I'm pretty convinced that input modules are not a good approach, but it's worth noting that:
- you should not need a dummy input anymore (see the modules in PYME.recipes.simulation for example) - the >= 1 input per module requirement was dropped about 15 months ago.
- you should use unifiedIO to get the files

csoeller, 2024-06-17:
Hi David,

Thanks for the quick reply. I fully agree, the recipes were created with the idea to encapsulate a set of operations. We currently have the use case that we have needed a solution for and the "hack" helps us to get something that works for now, even if not intended this way originally (and having some downsides the way it is done).

However, if it were possible to get a more proper solution along the lines you suggest above we would gladly support the development and contribute to it. The `save session` approach is something we do like and, for example, the visualisation software `paraview` supports something like that in a nice way by saving as `.pvsm` files- this is where we have used such functionality to good effect.

The command line idea with named datasources for visgui is a nice first step that could be an excellent starting point.

csoeller, 2024-07-08:
While the PR is still in the works, we leave this issue open, but should close the moment this is accepted into the main dist.