python-microscopy/python-microscopy #1527: [SUPPORT] How achieve compatibility with latest matplotlib?

csoeller, 2024-07-08:
Clearly, matplotlib, as most major Python packages, is a moving target. We noticed a first issue with 3.6.x which is quite easy to fix, see #1525 and #1526.

More recent matplotlib packages, i.e. 3.7.x and 3.8.x have a couple of changes that give rise to issues with `PYME/DSView/modules/graphViewPanel.py`. Notably,

 -  in matplotlib 3.7.X onwards `backend_wx.cursord dictionaries` are [removed](https://matplotlib.org/stable/api/prev_api_changes/api_changes_3.7.0.html#cursord-in-gtk-and-wx-backends);
 -  3.8.X [removes](https://matplotlib.org/stable/api/prev_api_changes/api_changes_3.7.0.html#cursord-in-gtk-and-wx-backends) `error_msg_wx` function in `backend_wx`

I am not very familiar what these do and how they are used.

**Question**: Does it make sense to make PYME compatible with more recent matplotlibs (I imagine it would)? If so, how can we best replace/alter the existing `graphViewPanel` implementation?

David-Baddeley, 2024-07-09:
We should certainly try to keep up with reasonably recent matplotlib.

A bunch of the stuff in `graphViewPanel` is copied from the internal code of an older version of wx (and hence uses a bunch of undocumented internal apis, which is probably why it has broken). We could update `graphViewPanel` so that it works with more recent versions, but might also want to consider if we need to continue support for `graphViewPanel` -  or if we might be able to make do with a lighter weight alternative which avoids re-implementing parts of the matplotlib internals. As far as I remember it's not widely used - potentially only with line profiles and spectrometers.

David-Baddeley, 2024-07-09:
Looking a little more, these are also used in the PSF and OTF plots, with the code being duplicated. The offending code is trying to encapsulate the toolbar functionality (including zoom etc ...) that you get in a standalone matplotlib plot window, but within our frame.

csoeller, 2024-09-11:
I have recently tested building PYME with python 3.11 and 3.12. At least conda-forge does not support `matplotlib<=3.6` conda install any more with these python versions (one could probably pip install 3.6, I imagine). This suggests to me we should try to loose the dependencies in graphViewPanel etc pretty soon to allow easing the matplotlib version constraints.

I looked at the matplotlib documentation and they do talk about embedding plotting with toolbars into wx using presumably defined (and therefore more stable across versions?) API calls, e.g. [Embedding in wx #4 ](https://matplotlib.org/stable/gallery/user_interfaces/embedding_in_wx4_sgskip.html) etc. Could this constitute a more robust route or do we need more functionaility than that?

David-Baddeley, 2024-09-11:
having a look, I think we can probably remove the `set_cursor` function completely  from graphViewPanel and just rely on the base class implementation ...

We can't just use `NavigationToolbar`, as we need our toolbar to be an aui toolbar rather than a regular one. I'm pretty sure that `MyNavigationToolbar` is just a copy of the code in `NavigationToolbar`.

csoeller, 2024-09-12:
Hi @David-Baddeley, many thanks for merging your PR #1546. This now allows building/running with matplotlib 3.8 provided the following small patch is applied to remove a couple of stale imports:

```diff
diff --git a/PYME/DSView/modules/graphViewPanel.py b/PYME/DSView/modules/graphViewPanel.py
index 3d27b0dc2..98196ed9b 100644
--- a/PYME/DSView/modules/graphViewPanel.py
+++ b/PYME/DSView/modules/graphViewPanel.py
@@ -27,7 +27,7 @@ import matplotlib
 matplotlib.use('WXAgg')
 
 from matplotlib.backends.backend_wxagg import FigureCanvasWxAgg as FigureCanvas
-from matplotlib.backends.backend_wx import _load_bitmap, error_msg_wx, cursord
+from matplotlib.backends.backend_wx import _load_bitmap
 #from matplotlib.backends.backend_wxagg import NavigationToolbar2WxAgg
 from matplotlib.backend_bases import NavigationToolbar2
 from matplotlib.figure import Figure
```

csoeller, 2024-09-12:
I can further report that with the patch above everything seems to work and build ok with `python=3.11` and `matplotlib=3.8` (subject to more proper testing at some stage, I have now a test conda environment for it).

A further issue was revealed when using latest matplotlib, really not a matplotlib issue but rather this allowing install of `numpy 2.X`. Here some `.pyx` and `.c` compiles fail since the numpy cython type interface seems to have changed in subtle ways. This is presumably fixable without too much hassle but I am not sure if numpy 2.X poses additional issues in other places (?).

Finally, when building with `python=3.12` one hits the removal of `numpy.distutils` with `3.12`, see also [here](https://numpy.org/doc/stable/reference/distutils_status_migration.html#distutils-status-migration). This sounds like a reasonable can of worms...

Just mentioning these here to flag what will likely come up in future.

David-Baddeley, 2024-09-12:
missing the import is a bit embarrassing ... 

`numpy.distutils` deprecation is going to be a much harder one to work around ...

csoeller, 2024-09-13:
With 3.11 `numpy.distutils` will be still supported until Oct 27. Most existing larger packages in the numpy et al. area seem to replace it with the `Meson` build system. Not a small job it seems...

More immediately relevant, how desirable is it to build and work with `numpy 2.X`? This can be established/tested with python 3.10 and 3.11 builds which otherwise seem to work ok (possibly earlier python versions as well but we at least have switched to 3.10 for good).