python-microscopy/python-microscopy #1558: [ENHANCEMENT] denoise point cloud data based on local density

ppa007hot, 2024-11-25:
In Vutara, it has a nice feature to denoise point cloud data based on local density:
[image]

Basically, it calculates the nearest neighbors for each point and generates a histogram of each channel and you can filter out sparse localization points (highly likely background noise).

[image]
[image]

This is super helpful when I want to generate iso-surface: the denoised data is less likely to have small non-connected surfaces.

Is it possible to implement this feature in future releases of PYME-Visualize? 

Thank you!

barentine, 2024-11-25:
@zacsimile is probably better at fielding this, so apologies in advance if I miss something obvious, but I'll have a quick go:
I'll describe something that's already in PYME and maybe you can let us know if that already does what you are looking for? 
1.  You can already estimate the local density for each localization in PYME. From the GUI, go to `View > Estimate Density`.
2. Run that, which will add a datasource to your pipeline recipe, by default it'll be called `dense0`.
3.  Select `dn` in the color-on menu if you want to color each localization by the local density estimate and get a feel for where to later threshold on this density estimate.
4. Click the `Pipeline Recipe` tab, and along the bottom you'll see a button that says `Add Module`, click it, and search or scroll down to `TableFilter` which you'll select and add to the recipe.
5. filter on `dn`
6. Make sure to update the Data Pipeline `output` (top left) points to your filter output, which I named `density_filtered`.
7. Now you can run your isosurface from the Mesh menu and generate it off a density filtered set of localizations.

Here is what your pipeline recipe might look like before the isosurface step, in case you want to copy, paste, and `Apply Text Changes` from the pipeline recipe tab.

```
- localisations.AddPipelineDerivedVars:
    inputEvents: ''
    inputFitResults: FitResults
    outputEventMaps: event_maps
    outputLocalizations: Localizations
- localisations.ProcessColour:
    input: Localizations
    output: colour_mapped
- tablefilters.FilterTable:
    filters:
      error_x:
      - 0
      - 30
    inputName: colour_mapped
    outputName: filtered_localizations
- pointcloud.LocalPointDensity:
    input: filtered_localizations
    input_sample_locations: ''
    output: dense0
- tablefilters.FilterTable:
    filters:
      dn:
      - 0.0
      - 0.01
    inputName: dense0
    outputName: density_filtered

```

[image]

[image]

David-Baddeley, 2024-11-25:
This needs some careful thought.

While I can certainly see a benefit of this type of denoising when creating an isosurface of certain structures (if there is, for example, prior knowledge that the label wouldn't appear on small structures such as vesicles), I have some reservations about making it too easy lest it become a standard part of workflows without carefully thinking about the implications. Specifically I think this kind of denoising is inappropriate when producing intensity images for visualisation (especially in papers). Because it modifies (suppresses) image information based on the local structure, it falls into the grey area of non-uniform image manipulation. It's not nearly as clear cut as, e.g. gel image manipulation, but it's easier to just not do it. For visualisation purposes, I think you are also suppressing useful information - although what you remove will almost always be non-specific labelling - e.g. single secondary antibodies, it is still useful to be able to see how much non-specific labelling there is.

When it comes to quantification, however, suppressing this non-specific noise will almost always improve the quantification accuracy. It also makes e.g. shrinkwrapping work a lot better. For isosurfaces used for *visualisation* (rather then quantification), there is still an argument that it can be useful to see the non-specific labelling, although this is less clear cut as it can make things really cluttered. Whatever you do do needs to be really well described in the methods (which is a strong argument for having to put together the steps to make it happen). 

@barentine has beat me to it with the fact that you can already do something very similar in PYMEVis and an explanation of how to do it. There is also a GUI method, although the recipe approach @barentine detailed is likely cleaner.

David-Baddeley, 2024-11-25:
Actually, GUI mode is not so bad ...

1) `View->Estimate density`
2) Open up the *Output Filter*, right click to add a filter, select `dn` and set the bounds on density

[image]

[image]

To assist with setting the density cutoff, after adding the filter, double click on `dn` to open the editor. This displays a histogram editor.

[image]

- click in the graph and hit `m` to set the scale to min-max
[image]

- remember the max value, and drag the right red bar to the left so you can see the dip in the histogram between the main peak and the zero (background) peak 

[image]

- drag the left hand red bar into the centre of the dip (this is analogous to what the ISODATA thresholding algorithm does)
[image]

- use the text box to set the upper bound to the max you remembered earlier (or anything larger)
[image]

The method of putting the lower cutoff in the middle of the dip is the easiest to justify as an unbiased way of separating specific and non-specific density histograms, but in practice you might want to set it a bit higher.

David-Baddeley, 2024-11-25:
Also to follow up on @barentines recipe method, the lower bound needs to be higher than 0 (or nothing will be rejected). I suspect he set this, but the value got truncated in display.

ppa007hot, 2024-11-28:
This is great! Thank you both for the help. I only use it for isosurface, not for quantification. I certainly will describe in the methods.