python-microscopy/python-microscopy #1562: [BUG] `setup.py develop` fails for for `setuptools>=65`

zacsimile, 2025-01-13:
With newer versions of `setuptools`, I get the error

```
Traceback (most recent call last):
  File "setup.py", line 64, in <module>
    from numpy.distutils.core import setup
  File "/Users/zachcm/miniforge3/envs/pyme/lib/python3.8/site-packages/numpy/distutils/core.py", line 24, in <module>
    from numpy.distutils.command import config, config_compiler, \
  File "/Users/zachcm/miniforge3/envs/pyme/lib/python3.8/site-packages/numpy/distutils/command/config.py", line 19, in <module>
    from numpy.distutils.mingw32ccompiler import generate_manifest
  File "/Users/zachcm/miniforge3/envs/pyme/lib/python3.8/site-packages/numpy/distutils/mingw32ccompiler.py", line 28, in <module>
    from distutils.msvccompiler import get_build_version as get_build_msvc_version
ModuleNotFoundError: No module named 'distutils.msvccompiler'
```

There's a discussion of the source of this error at https://github.com/pypa/setuptools/pull/3505. 

Installing `setuptools<65` solves the issue. Perhaps we should pin this from now on?

David-Baddeley, 2025-01-13:
Thanks for spotting this (and working out the correct pin). Between this and the pending `numpy.distutils` deprecation we have a lot of pain headed our way ... I've made some early attempts to switch to `meson` (for now just on `pymecompress`, and not quite working), but that is likely the way forward.

David-Baddeley, 2025-02-04:
numpy reccomends pinning `setuptools<60.0` ...

https://numpy.org/doc/2.1/reference/distutils_status_migration.html

zacsimile, 2025-02-15:
Why `meson` over something higher-level like `poetry` or `uv`?

zacsimile, 2025-02-15:
Ah, never mind, you mean as an alternative to `setuptools`. Sorry.

David-Baddeley, 2025-02-15:
Considering meson as a replacement for `numpy.distutils`, rather than `setuptools`. It's the only thing which looks like it might be able to give a reasonably clean replacement for our hierarchal `setup.py` files, and have the possibility of compiling for runtime SIMD dispatch.

csoeller, 2025-02-28:
Not sure if relevant, I have been pinning to `setuptools<=73` for a while now and that seems to work fine for all purposes - even though it emits a good pile of warnings when using the "old" setup commands.

David-Baddeley, 2025-03-03:
Making the pin somewhat more permissive would be great to get things
compiling for more recent python (I think <60 limits us to python <3.11).

csoeller, 2025-03-04:
I can confirm that building with python 3.11 works with latest patches for a few newer packages (traits 7.X fixes etc) on mac and win using `PYME-test-env` with a command like

```
python mk_pyme_env.py --python=3.11 --suffix=_1 -c mamba --use-git --pyme-repo "csoeller/python-microscopy"  --pyme-branch working-jan-2025
```

First impressions are that most if not all things work. This involves pinning setuptools(<=73), matplotlib(<=3.8), numpy(<2). Setuptools pinning is obvious, matplot and numpy would ideally be unpinned at some stage but require a few code tweaks, see #1527.

Python 3.12 builds invariably hit the missing `numpy.distutils`...

David-Baddeley, 2026-03-06:
@csoeller - Is this still an issue now we are using meson?

closed 2026-10-29 by David-Baddeley