python-microscopy/python-microscopy #1563: [ENHANCEMENT] PYMEAcquire UI improvements

barentine, 2025-01-21:
**Is your feature request related to a problem? Please describe.**
Running another instrument in PYME, used by an experienced microscopist other than me, who has correctly pointed out some frustrations with the user interface that we could stand to improve.

**Describe the solution you'd like**
- [ ] Ability to type in a z position for the positioning UI (currently can type in X and Y values, but Z goes to a slider). Slider for Z is fine, but would be nice to be able to click-into the text value and enter a new position similar to the slider labels worked out for PYMEVis view cropping.
- [ ] UI for setting an end condition to an acquisition without using a protocol or the action queue. I.e., run a time series for 1,000 frames, or 100 seconds, rather than either doing a single-shot acq, or starting and manually clicking to stop a time series. 

Will possibly update to include other notes in the coming weeks

David-Baddeley, 2025-02-27:
Sorry for not commenting earlier - was away when this came in and it fell off the radar. Most of these sound sensible. 
 - **z position:** I'm sure this has had an associated text edit box at some point in the past (might have gone away in a de-cluttering effort). Shouldn't be to hard to add back, although I am slightly wary of additional clutter - the UI is already super busy. In the interim, might the `focusKeys` plugin (see the default simulator init file) be useful? Should the future text-entry be in the main UI, or accessible from a menu?
 - **max_frames:** With the revamped actions and analysis chains, it might make sense to bring max frames out into the *Time/Blinking series* controls and drop it from the *Queued Actions*. A fixed length XYZTC time series mode is also on the cards (potentially allowing timelapse - i.e. decoupling frame rate and integration time.
 - **Entering ROI coordinates**: Suggest putting this under the *Camera* menu, along with *Toggle ROI*, and pop up a dialog. Don't really think this needs to be shown all the time.
 - **Keeping manually entered filename stub:** Probably makes sense, but avoided to date for a couple of reasons:
     - To try and incentivise putting sample information in the metadata, rather than the filename so that it is in a consistent schema, can be ingested into a database, and is searchable. In my opinion putting sample information into the filename is a bit of an anti-pattern as it is invariably incomprehensible to anyone other than the person who took the data.  On many occasions I've been tempted to actually remove the edit field entirely.
     - To maintain directory sort order, and ensure that suffixes can be linked back to acquisition order. 
     - The complexity of working out if a file already has a suffix, and how to increment if you have arbitrary filenames. A simplistic solution would generate filenames which looked like *custom_filename.hdf*,  *custom_filename_B.hdf*, *custom_filename_B_B.hdf*, *custom_filename_B_B_B.hdf*, etc ... If we did change it, the simplest approach would likely be take the suffix out of the editable bit, so you could only ever edit the stub.
  - **Centreing view on  selection:** Not silly, but not completely trivial to implement with how the view is currently structured. You'd also need to, e.g., do some magic to make sure things do something sensible when the window is resized. Might cause some unexpected behaviour (once you are at a point where the view scrolls, it might make more sense to make the centre of the zoomed view match the centre of the current view, rather than do it based on the selection). Might all go away if we roll with the OpenGL accelerated image view across everything (which is definitely on the cards - to reduce maintenance cost if nothing else).

David-Baddeley, 2025-02-27:
A menu-based GOTO(x,y,z) dialog might also be a way to reduce the frustration with the current (x,y) positioning - because the controls are used as readout as well as entry, you can't enter both an x and a y position and then hit go, because the control will update to the current position as soon as it looses focus.

Maybe if we changed the UI so that the x and y had static text controls that read out the current position, along with the nudge controls, and then moved the *GO* button either to the right of or beneath everything (z included), and made it pop up a small modal for a 3D GOTO. @barentine - does this sound acceptable?

David-Baddeley, 2025-02-27:
Tentative postioning panel changes:

[image]

With the future option to, e.g. have a list of bookmarked positions to choose from in the dialog.

barentine, 2025-02-27:
I think the dialog for text entry would go a long ways towards usability! Particularly since I always seem to have some focus-escape issues on the current XY. Thanks for having a go at this, David! 
Could consider MOVE instead of GOTO for the button label.
Might also be nice to let folks toggle whether Z gets a slider or the same UI as XY.

barentine, 2025-03-31:
Any thoughts on adding a frameWrangler control menu somewhere in campanels @David-Baddeley ? Possibly paired with making a camera acquisition mode panel a part of the base cam panel? Or, probably better from a user perspective, integrating the two into the camera menu? 

I am suffering a bit from the implementation of single-shot mode, being single-shot for spooling purposes, but effectively continuous running from the perspective of framewrangler (coutesty of the if not contMod, need exposure start logic). If I change the camera over to single-shot mode, then spool a single-shot series, afterwards the camera keeps getting re-fired repeatedly, which is not a behavior I am after with my stage-scanning. 

I wonder if we could have camera acquisition control modes and framewrangler mesh a bit cleaner. For example, if I have a camera in software trigger mode, rather than single-shot mode, I might want to have framewrangler re-triggger so it acts ~continuous when not spooling. Or, my current predicament, I might want the camera to not be re-triggered when not spooling, but act single-shot when acquiring. 

I think folks would appreciate a play/pause button that handled pa.Prepare()/pa.start() and pa.stop(), respectively.
Maybe in addition to a dro-down menu to select camera acquisition modes for the current camera, with an 'auto re-trigger' checkbox which greyed out for continuous & hardware-trigger modes, but let users toggle for software trigger and single-shot?

David-Baddeley, 2025-04-01:
Can I assume this is your custom camera shim and that you use the eye pieces to find your position for your stage scanning acquisitions and don't really want / use a live preview? What does your camera shim do if it's told to run in continuous mode?

I'm just trying to firm up the use-case for what a non-continuous / not re-triggered live preview mode might look like and when it would be used ...

barentine, 2025-04-07:
Hi David, thanks much for the info and consideration! Somehow I missed your replies - sorry for the delay.

Yes, I am actually using a motorized flipper mirror to switch detection between an IDS CMOS and my point-scanning detector. 
I have the pointscan camera set to repeatedly trigger scans in 'continuous' mode. 

I would like to be able to find an ROI by click-dragging the stage around with the CMOS active, then switch cameras, and not have the thing start scanning unless I click 'start' in the spool panel. Even analog-driven, the scanning can take awhile, and I don't immediately have a way to interrupt a scan if we're part-way through and then decide we want to start spooling a new frame or something, which is maybe a confusing way of saying it is nice to be able to not have the stage moving until I really want to acquire a frame.  

In one sense, I'm after an option to separate OnExpReady / framewrangler polling from triggering my camera once it is 'primed' for a single-shot acquisition. One hack for that would be to give my pointscanning camera a contMode property that was always True even if it is very much not in continuous mode. In another sense though, I think I'm after a way to pause the framewrangler outside of an acquisition, so that I can e.g. set to continous mode, run a protocol, and when I click 'stop' after some frames actually have the stage stop moving too. Might be a niche use case, but I thought it worth sharing because the non-cont mode magic did trip me up a bit!

David-Baddeley, 2025-04-10:
Sorry about the non-timely feedback on this one. At the moment I'm pretty much limited to Fridays for PYME stuff. To me it sounds like we need a camera-dependent option for what to do when "idle" (i.e. not in an acquisition). For CCD cameras, this could default to the current continuous mode. For the point scanning "camera" it should probably just be to do nothing. Will think a bit about what this means for the frame wrangler etc ... 

Quick thoughts ...
- Logical place to switch between "acquisition" and "idle" modes might be in `SpoolController.SpoolStopped()`
    - Is this guaranteed to get called on abort?
- Would need matching mode change call in start
- If we end up stopping the framewrangler for "idle" with a camera which doesn't want continuous running (i.e. point scanner), we'd need to restart it when changing cameras
- Would need to decide if it makes sense for the core "idle" and "acquiring" state logic to reside in the `Microscope` object  or `SpoolController` (leaning towards having it in `Microscope` as this means that switching cameras doesn't need to talk to the `SpoolController` as that would seem inefficient.

barentine, 2025-04-11:
No problem on delays, David! And excellent points. "Idle" is really what I was trying to get it - glad you put your finger on it. A hook for going into and out of idle mode would be great to have for e.g. returning a stage scanner to center position, shuttering lasers, etc. At the moment, the switching camera logic seems to try calling SetShutter open on the way in, but not closing the shutter on the way out? Some flexibility with hooks for shuttering/unshuttering cameras/lasers/whatever would be neat there.