python-microscopy/python-microscopy #1564: [BUG] wx DrawRectangle, DrawLine require ints, but ViewPanel get_coords returns floats

barentine, 2025-01-21:
**Describe the bug**
We have a couple of spots where we pass floats wrapped in ints into calls to wxPython display contexts for drawing lines, or rectangles, etc. We have to cast to int to avoid hard errors on later version of wxPython, eg 4.2.1.

[image]

Problem is that most of the time we are getting these coordinates from e.g. PYME/DSView/arrayViewPanel.py `pixel_to_screen_coordinates` which returns float. 

So, I guess what I'm wondering is whether @David-Baddeley would prefer a PR wrapping the rest of the DrawRectangle / DrawLine args with int casts or whether it makes more sense to just return screen positions in integer pixels in the first place. Maybe there's an issue with the later approach, I haven't looked into it much, but I wanted to check before submitting a bunch of int-casts in a PR. I'm certainly fine with either approach

**To Reproduce**
Steps to reproduce the behavior:
1. Open an image stack in PYMEImage
2. add PSF extraction tools module
3. try and tag a PSF
4. See error drawing tagged rectangle

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] win10 or 11
 - Python version: 3.10
 - numpy version: 1.26
 - wxpython version [if a UI bug]: 4.2.1

David-Baddeley, 2025-01-22:
Hi Andrew,

I'd opt for keeping `pixel_to_screen_coordinates as float, as the mid term
plan has the view layer moving to OpenGL, which uses float coordinates (and
on high-dpi screens might actually do something useful with fractional
pixels).

Best wishes,
David

barentine, 2025-01-22:
Sounds good - thanks, David!