python-microscopy/python-microscopy #1574: [SUPPORT] software ROI cropping for cameras that don't allow flexible ROI setting

barentine, 2025-02-26:
One of the painful things with the IDS cameras is that some of them do not offer much flexibility in terms of setting ROIs. We would ideally set a much smaller ROI than is supported by our camera - otherwise we save loads of extra data, and there is also the annoying aspect of having to find the region you care about on the chip when PYMEAcquire loads up with a relatively massive ROI compared to the region you are actually interested in.

**Where are you stuck?**
My familiarity with the `Acquire.Hardware.Camera.MultiviewCameraMixin` makes it tempting to set up that way and have a defined ROI center I can adjust the size of regardless of the actual chip setting. I could see an argument even for adding a method for editing/shifting the ROI center.

However, MultiViewCameraMixin is now deprecated with a TODO to remove it. Is there already a replacement for my use case? I've had a quick look at `Acquire.Hardware.splitter`, but it does not seem to be particularly useful for a live Software cropped ROI during acquisition.

What makes sense here? I see how the Multiview and Splitter stuff replicate the 'multi' aspect, but I do think it would be nice to allow a software crop so that in PYMEAcquire, you could actually get the ROI you try to set even if your camera has some finicky stuff going on (which, so far, I think they all have _some_ level of stride enforcement, etc.). Is there a refactor option where we keep the software ROI setting aspect of Multiview, but remove the Multi in deference to splitter? Otherwise, I think I will try adding a software ROI setting to the `ids_peak_cam.py` so if you try to set an ROI and the hardware ROI can't go far enough, `ids_peak_cam.IDS_Camera.ExtractColor` will crop further so you're only fed the ROI you asked for (modulo asking for a smaller ROI than you could get, rather than a larger one).

David-Baddeley, 2025-02-27:
Hi Andrew,

the deprecation on `MultiviewCameraMixin` is specifically about the copy in Camera.py - I wanted to de-clutter Camera.py and moved the functionality into it's own module, `PYME.Acquire.Hardware.multiview'. This is noted a couple of lines down from the comment (line 979).

I also changed to be a wrapper, rather than a mixin `MultiviewWrapper` which means you don't need to subclass your camera when you use it, so we don't end up having a proliferation of `MultiviewHamamatsu`, `MultiviewUEye` etc ... classes which are almost identical. With the wrapper, you can enable multiview on any camera with a fairly subtle changes to your init script (you create your camera object as normal, and then create the wrapper around it, anything non-ROI related just gets passed through to the base class).

There is an example of usage in `init_sim_htsms.py`.

David-Baddeley, 2025-02-27:
That said, if you don't specifically need the multiple ROIs and toggling etc, it might be simpler to have a stripped down `SoftwareROIWrapper` which handled choosing an optimal hardware ROI and doing the software cropping to give you an effective ROI (this would eliminate a lot of the complexity and metadata / image origin issues that come with the switchable channels in the multiview implementation).